qidi-community/Plus4-Wiki #10: Motherboard Fan

kuehmayer, 2024-10-23:
I installed the sunon fan (exactly same as the link, just from amazon) and noticed it only really kicks in at ~99% and struggles to turn below that. Tested it with an external power supply and worked fine. The sunon states 1.2W (0.05A), I measured 0.044A max, and it is starting at 3V (0.012A). For the original it states 0.1A, however, I measured  0.023A max, and it´s starting at 12V ( 0,011A). 

So I think the output current to overcome the start up for the sunon is not high enough from the 4 Plus. 
Is it working for anyone else or am I doing something wrong?

For now, I´ll adjust the PID settings to start at 0.998 and max 1 haha.

spooknik, 2024-10-24:
Hello! 

I think you ended up getting a version of the fan that does not have the ability to do PWM. So it's just either fully on or fully off... can you still return it to Amazon and get something else? [These fans from Anvison](https://www.amazon.com/ANVISION-2-Pack-Brushless-Cooling-Bearing/dp/B0C18G953Z) should do the trick if you're buying from Amazon.

stew675, 2024-10-26:
The Anvision ones that @spooknik linked to are exactly the ones that I use, and I can confirm that they work with the profile as listed.

closed 2024-10-31 by spooknik