qidi-community/Plus4-Wiki #9: Screw Sizes?

nicwilson58, 2024-10-23:
Sorry if this is the wrong place to ask this.   the video for the Offical QIDI Toolhead fix video states to use 3*8 self tapping screw.  What size is this,  all the screw sizing in Australia does come in that format so the size of the two screws required escapes me.

thanks
Nic

stew675, 2024-10-23:
I'm fairly certain that they mean any self tapping screw of at least 3mm in diameter, and of 8mm in length.

I think the guy in the video just used standard M3x8mm SHCS screws, and not actually self-tappers.

A 6g screw is 3.5mm in diameter, so that would work too.  I personally used 6g screws myself, of 3/8" (~9.5mm) in length.

Try to use screws that have a flat tip, instead of a sharp pointy tip, or at least file down the tip of any screw you use to be flat.

Since you're in Australia, something like this: https://www.bunnings.com.au/zenith-6g-x-12mm-zinc-plated-pan-head-sheet-metal-screws-100-pack_p2437506

but you only need 2, and file the tips down to be fairly blunt.

I hope that helps.

nicwilson58, 2024-10-23:
thank you for that I found some black,  M3 x 8mm flat screws in my 3d printer kit that look the same as the ones in the video so they should work fine hopefully.