qidi-community/Plus4-Wiki #12: Stepper Motor Tweaks Considerations

nicwilson58, 2024-10-28:
When I read the "Stepper Motor Tweaks" changes it seemed like a great idea.  I purchased a very good quality 80mm fan with the same connector so no wiring changes.  I printed the new mount and fitted it.  I then made the changes as recommeded in the "Stepper Motor Tweaks" guide.

Before making the changesand when idle and no board fan running fan running the "host" temperature ranged from 65 - 75c,  so I could see why the fan should be enabled always at a lower speed,  and why you targeted 45c for the host sensor.

After the changes I found the 80mm fan running at 30% could very easily bring the host temperature down to 45c,  and even while printing the host temperature never get above 45c and so the board fan remains at 30% and rarely gets faster and the stepper drivers were easily reaching above 85c (with the new 1.15 current setting)

I made some alterations to check what the stepper temperatures would be  at 100% fan printing the same model.  The stepper temperatures were now between 65 - 69c,  far cooler than before.  I understand we cannot use the TX2240 temp sensor to control this, but to me so far it seems the host temperature is also not very suitable.

Of course the findings are for my printer,  others may be different,  buy I am interested in your thoughts around this.
Regards
Nic

jojolll, 2024-11-05:
Hi,

I also agree with you about the cooling of the steppers, which I found insufficient with these tweaks. I kept the original logic where the fan runs at 100% as soon as the steppers are active (and in a few other conditions).

I'm considering adding an idle mode soon, which would leave the fan running at 25%, ensuring that the card still gets cool air even when idle. The steppers, even when off, aren't completely cold, for example, and the CPU heats up too. So, switching off the fan isn't always optimal.

nicwilson58, 2024-11-05:
> Hi,
>
> I also agree with you about the cooling of the steppers, which I found insufficient with these tweaks. I kept the original logic where the fan runs at 100% as soon as the steppers are active [...]

I have already done this,  and found some things.  Many fans I tried are not stable at 25%,  so at idle mine are at 35% which keeps the chipset cool.  Idle does not seem stable also,  the idle timeout is sometimes very rarely called so I dont use it.  use the mcu temp to reduce the fan to 35% when the target temp of 50c is reached,  it then idles at 35% and stays cool below 50c, 
 In the PRINT_START I alter the target to 30c which it can never reach but causes a 100% fan while it tries thereby keeping the steppers cool.  On print end or print cancel,  the target temp is returned to 50c, slowing it down again.  It works very well, I can supply my printer.cfg if you want.

nicwilson58, 2024-11-09:
> Yes, I want your fan config :)

**Sorry for the delay was away for work,  Add it to you printer config like this:

Comment out original fan config  and add  new fan config which has target temp of 50c min speed of 50% these can be edited, its looking at the mcu temp,  fan will go 100% until mcu reaches 50c which will them drop it to 50%,  See furhter down for next parts.**

#[controller_fan board_fan]
#pin:U_1:PC4
#max_power:1.0
#shutdown_speed:1.0
#cycle_time:0.01
#fan_speed: 1.0
#heater:chamber
#stepper:stepper_x,stepper_y

[temperature_fan board_fan]
pin:U_1:PC4
max_power: 1.0
shutdown_speed: 1.0
cycle_time: 0.01
off_below: 0
sensor_type: temperature_host
control: pid
pid_deriv_time: 2.0
pid_Kp: 5
pid_Ki: 2
pid_Kd: 5
target_temp: 50
min_speed: 0.50
max_speed: 1.0
min_temp: 0
max_temp: 100

**To increase fan from 50% to 100% override the 50c target temp with a temp it can never reach but goes to 100% trying,  do this by adding this to PRINT_START**

SET_TEMPERATURE_FAN_TARGET TEMPERATURE_FAN=board_fan TARGET=30

On PRINT_END  &  CANCEL_PRINT  set back to 50c target temp by adding this:
SET_TEMPERATURE_FAN_TARGET TEMPERATURE_FAN=board_fan TARGET=50

**note I did not use idle timeout as for some reason it was never called.**
let me know how you go, it works great for me.

closed 2024-12-04 by stew675