qidi-community/Plus4-Wiki #16: Recommended Gdstime fan is not really better

m4dm4n, 2024-11-12:
Comparing the specs between default heatsink fan and Gdstime, it's evident that the default one is better performer and quieter. So I don't really get why the other one i recommended. 

Ruiapple RA-DC3010 24M
Current : 0.08A
Speed: 10000RPM
AirPressure: 4.2 CFM
Noise: 32dB

Gdstime GDA3010-5F 24V
Current : 0.06A
Speed: 12000RPM
AirFlow: 4.1 CFM
Noise: 35dB

spooknik, 2024-11-14:
Myself and @stew675 checked the model numbers of our stock fans which were RA-DC3010 M, which is only rated for 3.8CFM. I could be Qidi started shipping the more powerful version RA-DC3010 H which is on par with the GSDtime 3010. If you have a different model of fan stock on your printer, let us know. 

[image]

https://www.fanscooling.com/dc-axial-fans/ra-dc3010.html

m4dm4n, 2024-11-14:
Granted, but I would always look at the official vendor specifications when comparing products:

https://www.ruiapple.cn/Products/DC-Axial-Fans/450.html

https://www.ruiapple.cn/uploads/soft/210810/3010.pdf

Yes, it's still the same model, but the specs on their page are a little different.

spooknik, 2024-11-15:
I don't have an accurate way to measure fan CFM, but the GSDtime fans push a lot more air than the stock fan even though the CFM ratings are close on paper. You can tell just by putting your hand on the other side of the hotend when the GSDtime fan is installed. It wouldn't be the first time specs on components are 'inflated'. 

Either way, it seems the stock fan can keep up with cooling the cold side of the hotend with the newer ceramic heatbreak. So changing out the fan isn't as critical as it was for us on the initial units. 

GSDtime isn't an A-tier supplier of fans, Sunon or Delta are better, but don't make 24V fans in 3010 size. So the GSDtime recommendation will stick around.

closed 2024-11-15 by spooknik