qidi-community/Plus4-Wiki #18: Inconsistent CN links for replacement hotend heatsink fan

jaydenseric, 2024-11-19:
There are 2 CN links for a replacement hotend heatsink fan that I've discovered so far, and both are different:

1. https://github.com/qidi-community/Plus4-Wiki/blob/9aa1d78b20cca1841f9ef52bfa6b111f09b86f95/README.md?plain=1#L90
2. https://github.com/qidi-community/Plus4-Wiki/blob/9aa1d78b20cca1841f9ef52bfa6b111f09b86f95/content/nozzle-clogging-tips/README.md?plain=1#L28

The second one doesn't select a particular variation of 4 product options, which is correct?

What is the best link to purchase? Let's make them consistent in all locations.

stew675, 2025-01-29:
You are correct.  The AliExpress item does not match the other two, and in fact it appears to be changing over time.  I will remove that link.  I would suggest finding a GDSTime fan that matches the specs given on the EU shop page.  The US shop page is also correct, but it mistates the 12,000RPM as 1,200RPM, which is a typo.

stew675, 2025-01-29:
@spooknik I am unable to edit your file https://github.com/qidi-community/Plus4-Wiki/blob/9aa1d78b20cca1841f9ef52bfa6b111f09b86f95/content/nozzle-clogging-tips/README.md

Could you make the correction?

vega-d, 2025-02-18:
was fixed

closed 2025-02-18 by vega-d