qidi-community/Plus4-Wiki #33: Tweaking the meshbed accuracy changes for more speed while retaining tolerance and accuracy of probing

guykuo, 2025-02-08:
Thanks for the wonderful github on meshbed probing improvement. I have implemented those on my printer, but with some changes in values that might be helpful to other Plus 4 owners

As the github strives to achieve....

A. Increase Z stepper motor accuracy for both actual position and tracking of their position.

The Z and Z1 stepper motor microstepping is reduced to a more realistic 16. This increases their torque and improving how well their position tracking matches actual physical position. Interpolation is also turned off for those Z steppers to also make their motions more accurate. 

B. Improve bed mesh probing accuracy and tolerance to variability in inductance probes.

C. Maximize speed of overall bed mesh process by accelerating motions that are not actual probe sampling motions.

**** CAVEAT: Back up your printer.cfg file before making these changes. Realize you could be making settings changes that could be really bad for your printer if you make a mistake.

Once you have completed the edits, you will need to "save and restart" for them to take effect.

All the setting changes are done in printer.cfg. 

The following are my actual Plus 4 printer.cfg setting changes. Mine differ from the github suggestions.

I only list the settings that were changed. Leave the rest alone.

======= [stepper_z]

microsteps: 16

======= [stepper_z1]

microsteps: 16

======= [tmc2209 stepper_z]

interpolate: False

======= [tmc2209 stepper_z1]

interpolate: False

======= [smart_effector]

speed:3.25

lift_speed:18

samples: 1

sample_retract_dist: 6.5

samples_tolerance: 0.013

“Speed” is how fast the bed moves towards probe while the machine senses for bed. This speed originally 5 and github suggested 2.5 to improve accuracy. I tested a series of speeds and found 3.25 was a reasonable compromise that was slow enough to get repeatable readings, but not excessive prolong each probing motion.

“lift_speed” was increased from 5 to 18. “lift_speed” is how quickly the bed moves away from the probe. Measurements are not being done during the away motion and there is no reason this motion cannot be faster than the actual approach motion. I tried several increased lift_speed settings and kept it reasonable to not over stress the axis. A setting of 20 was particularly noisy. 18 was faster, but still quiet. Thus I settled on 18.

“samples” was changed to 1 AFTER doing tests with it at 2 samples / point to verify my speed and distance settings work for my probe.

“sample retract dist” is how far the bed moves away from probe when going after additional samples. Github suggests 10 mm, but because my probe was already OK with original 5 mm, I only increased it partially to improve tolerance to probe variability. If your probe is particulary odd, you may need to increase the distances out towards 10 mm. We want enough distance, but not so much that we excessively prolong probing motion.

“samples_tolerance” was tightened to 0.013 to help verify that distance and speed were working correctly. Tolerance is tested by machine ONLY if more than 1 sample / point specified. Once you have things verified, you can set samples down to 1 to nearly double speed, but verify settings with samples = 2 before dropping down to 1 sample.

=======

[bed_mesh]

speed: 400

horizontal_move_z: 6.5

“speed” is how fast x, y motions are performed during bed mesh. We have a fast core xy structure, so I boosted the motions to 400 from the stock 150.

“horizontal_move_z” is the baseline distance the bed is to be moved during probing. (If you change this value, you should also make “sample_retract_dist” in smart_effector match)

Github suggests 10 mm, but as noted in the smart_effector section, I compromised at 6.5 mm. My probe is probably a more typical one that is OK at 5 mm. Github suggests 10 mm to be tolerant of nearly all probes. I went with a shorter distance to get more speed while still tolerating more probe variability. If your particular probe needs more distance than 6.5 mm, you will notice probe values differ excessively during testing. In such case, increase the horizontal_move_z towards 10 mm. 

=======

[z_tilt]

horizontal_move_z: 10

retries: 5

retry_tolerance: 0.013

Z-tilt distance I went with Github suggestion of 10 mm because there is little speed penalty for longer distances compared to that during bedmesh. Retries and retry_tolerance were matched to those in smart_effector.

Unkleskunky, 2025-02-09:
Awesome!

I am an early adopter but highly inexperienced 3d FDM'r. Even with my
Printrbot maker edition I had very few problems, my qidi plus 4 is worlds
more capable but still I have very few issues. I tend to print pretty
simple things with pretty practical materials but I would like to evolve.

I would like to learn to be comfortable making these kind of edits, can you
recommend a good starting point to learn this type of thing?

Much appreciated,

US

Much appreciated.

guykuo, 2025-02-13:
There is likely another place in the printer software that also needs the higher accuracy z-probing longer distance and slower speed.

There is also a gcode.macro.cfg file that has at least one command involved in probing for setting z. This seems to use speed and distance settings that are independent of those we adjust in printer.cfg.

Notice the line I show in orange. I think we should also change the speed and values in this to get more z probing accuracy during NON-bedmesh probes.

probe probe_speed=5 lift_speed=5 samples=5 sample_retract_dist=5

probably should be longer distance and slower speeds like....

probe probe_speed=**2.5** lift_speed=5 samples=5 sample_retract_dist=**10**

**

[image]

**

guykuo, 2025-02-17:
More data to consider. I'm now of the opinion that given a well behaving inductance probe, the probing distance is a more important factor tha probing speed in how accurately the probe repeats measurements.

Using a fixed probing speed of 2.5 mm/sec, I looked for inconsistencies in z-probing measurements during Z-homing while varying initial distahce to nozzle.

[image]

That series of measurements showed  z travel during probing of 5 mm is just within acceptable for my probe. The suggest value of 10 mm probing distance is more than required to get enough leeway. My use of 6.5 mm probing travel is above the distance needed and faster than traveling 10 mm for probing.

Next, I investigate speed of probing. The results were astonishing. The sweet spot in terms of fastest probing speed without increase standard deviation was about 12 mm/sec. That is far faster than the 2.5 mm/sec suggested or the 3.6 I have been using.

[image]

Given those results I validated the speed of 12 mm/sec on my machine and it words. Even detected a piece of debris beneath my print sheet. I set toleraance to  0.025 which is a bit tigher than 1/5 of my thinnest layer height. Validate with 2 samples/point and once that passed I set smart effectos as below...

[smart_effector]
pin:U_1:PC1
recovery_time:0
x_offset: 25
y_offset: 1.3
z_offset: 0.000001
speed: 12
lift_speed:18
probe_accel:50
samples: 1
samples_result: submaxmin
sample_retract_dist: 6.5
samples_tolerance: 0.025
samples_tolerance_retries:5

Also set bed_mesh to even faster non-probing motions

[bed_mesh]
speed: 600
horizontal_move_z: 6.5
mesh_min:25,10
mesh_max:295,295
probe_count:9,9
algorithm:bicubic
bicubic_tension:0.4
mesh_pps: 2,2

It works on my machine, but I spent hours validating my particular induction sensor. 

Here is a full bed mesh being performed with these settings....
https://youtu.be/0GjVF6VDE3g

After that, had good success printing with a z-offset of zero (that is my usual z-offset for the particular machine specimen)

vega-d, 2025-02-18:
This looks very well researched and I agree, however the changes as you present them are not in a format that can be merged into the wiki. Please look at a guide for contributing to a github repo (for example, [this one](https://docs.github.com/en/get-started/exploring-projects-on-github/contributing-to-a-project?tool=webui)) and submit your changes as a pull request instead.

closed 2025-02-18 by vega-d

pravic, 2025-02-18:
> however the changes as you present them are not in a format that can be merged into the wiki.

And yet you closed this as "completed" which is quote misleading. Perhaps, it should be something else?

[image]