qidi-community/Plus4-Wiki #38: After SET_PRINT_STATS_INFO modification there are no layer infos (actual layer/total layers) and no slicer time on printer display and in Orca Slicer

GitHag42, 2025-02-12:
I updated to FW 1.6.0 and did some changes to the configuration as suggested here in the wiki.
With the new code for SET_PRINT_STATS_INFO from [here](https://github.com/qidi-community/Plus4-Wiki/tree/main/content/chamber-heater-issue#a-firmware-configuration-workaround) I have no actual layer and no slicer time shown on the printer display and in Orca Slicer (Device tab) while printing. There are just dahes (--). 

The code added at the top in gcode-macro.cfg is (as currently shown in the wiki article):

`
[gcode_macro SET_PRINT_STATS_INFO]
rename_existing: SET_PRINT_STATS_INFO_BASE
gcode:

    {% set curlayer =  params.CURRENT_LAYER|default(1)|int %}
    {% if (printer.toolhead.position.z) >= 268 %}
        # Set chamber target to 10C, which still keeps the chamber heater fan on
        # This allows the print bed to warm the chamber more effectively even
        # though the heater coils are effectively disabled
        M141 S10
    {% endif %}
    SET_PRINT_STATS_INFO_BASE CURRENT_LAYER={curlayer}
`
(I don't know how to put the hole macro in the code view here, so I added a new line after "gcode:" which is not present in my gcode-macro.cfg)

When I comment this code and restart the printer and start a new print the info is shown again.

Of course I followed the important [hint](https://github.com/qidi-community/Plus4-Wiki/tree/main/content/chamber-heater-issue#important-update-for-v160-firmware) for FW 1.6.0 and set target_min_temp: 10            #<--- Ensure that this field is set to 10C (default is 45C) in the [heater_generic chamber] config section.

Is the position where the code is added in gcode-macro.cfg important or am I missing something else here?

Thanks for any hints :-)

Ken6432, 2025-02-12:
For what it's worth, I have not made the referenced heater change to my 1.6 config, but seemingly randomly, I don't get the remaining time displayed during printing.  Sometimes it shows and sometimes it doesn't.  The issue may not be related to the changes made.

jwestenberg, 2025-02-18:
Could it be that it does not show layer information while using the "infill combination" setting?