qidi-community/Plus4-Wiki #45: Bottom X axis rail is self-lubricating bronze, warn against oiling

mnemnion, 2025-03-04:
The [video here](https://old.reddit.com/r/QidiTech3D/comments/1j282k8/qidi_plus_4_extruder_lower_bushing_repair/) shows clearly that the lower X rail uses a graphite-plugged bronze bushing.  These are self-lubricating and should not have any oil added.

There's some language in the [lubing rods and screws](https://github.com/qidi-community/Plus4-Wiki/tree/main/content/lube-rods-screws) part of the wiki which warns against using grease or PTFE, which is true, but machine oil shouldn't go in there either.  Nothing but graphite should be on a graphite-lubed bushing, and it provides its own so regular maintenance shouldn't involve adding more (although that would itself be harmless to the bushing). 

I think it would be better for that section to say something like this: "The bottom rod of the X axis uses a graphite plugged bronze bushing.  While the QidiTech maintenance video shows lube being applied to this rod, the graphite plugs provide the necessary lubrication and we believe adding oil is contraindicated.  If, over time, the bottom bushing squeaks, a small amount of graphite powder can be worked in, but this is likely to never happen".

I could be wrong here, maintaining shop tools is not my profession, but adding oil to a self-lubricating graphite bushing goes against everything I've learned and any reference on the Internet I can find. 

Somewhat more speculatively (in that I couldn't find a primary source for this), adding some graphite powder should adsorb away any machine oil already added due to following the advice on the official wiki.  This would be a couple of rounds of light powdering, moving the tool head to work it in, wiping the gunk off the rail, waiting awhile for adsorption to occur, and repeating until the graphite is dry.   I don't endorse adding that to the wiki until it's been tried, I'm convinced it's perfectly safe, and reasonably confident it will work.  But I'm not going to contaminate my bottom rail with machine oil just to try cleaning it up, so this is an untested hypothesis.

dssamusaran, 2025-03-04:
Very interresting topic. It enticed me to do some googling around and it seems you're right.
Bottom rail should not be oiled or greased. Only cleaned with a rag.

The fact is my machine was already factory greased even on the bottom rail. So after very few prints I saw black gunk on the rail.
I went on Qidi wiki and followed their guide that indeed told me to grease the bottom rail. Which I'm doing like an idiot every now and then...

Would this kind of product work to try and clear up the grease following your method?
https://www.amazon.com.be/-/en/ProGraphite-Micronised-Graphite-lubricant-conductivity/dp/B07C75XTF2

I've absolutely never used graphite powder so I'd like to make sure it's the correct thing. 
And I'll also wait for some more insight to see if the process seems good.

By the way, as of now I don't really have any black gunk buildup and the machine is working fine without squeaking.
Maybe I completely killed the graphite of my bushings?...

mnemnion, 2025-03-04:
> The fact is my machine was already factory greased even on the bottom rail.

It probably felt that way, yes. Might have been as well. I checked lubrication on all the moving parts before starting the machine, since if that step got skipped at the factory, it's bad news and I can easily prevent the problems by doing it myself.

It was only later in that thread, when `/u/pointclickfrown` pointed out the graphite plugs in the bushing, that I checked with a piece of paper towel, and that rod (only) is graphite lubricated.  The combination of graphite and machine oil does in fact look like black gunk, and it's not good.  It's not terrible either, but best to clean it out.

The product you linked to is the right kind of stuff, yes.  I wouldn't worry about the graphite wearing out, the bushings literally have [chunks of graphite added](https://www.bronzegleitlager.com/photo/ps35488774-self_lubricated_bronze_embedded_graphite_bushing_c86300_alloy.jpg) and should last at least as long as the toolhead.  You're probably fine if you just don't add more oil, but if you want to check, take white paper towel and gently rub the rod with it.  You should see smudged graphite, it looks exactly like smudge from a pencil because that's what pencil smudge is, the shade is quite distinctive.

That picture isn't the literal Plus 4 bushing btw, but it looks basically identical.

jojolll, 2025-03-14:
The sticker on the bottom of the machine specifically states that the bottom X axis should be lubricated with oil from time to time with graphite bushing, to avoid friction if it makes noise (which is never good).
I imagine that graphite alone is insufficient for the repeated and rapid friction of a 3D printer? Maybe that's the reason for the message?

I don't know what to make of that in terms of service life. I almost feel like continuing to add a drop of oil from time to time.

mrelwood, 2025-04-21:
> The sticker on the bottom of the machine specifically states that the bottom X axis should be lubricated with oil from time to time

Exactly. And there is a reason for the sticker being there.

We hunted for a solution with Qidi support for a month to bad print quality issues on a new Plus 4, which miraculously went away as soon as I cleaned up and oiled generously the lower X axis rod.

I get that a graphite bushing is used without lube in many instances, but a rapidly moving tool head doesn’t seem to be one of them.

stew675, 2025-07-15:
Closing this issue as it appears to have been resolved.  Good discussion above. @mrelwood's experiences echoes my own.  Oiling the lower graphite bearing with a *light* machine oil appears to solve issues with its grabbiness.