qidi-community/Plus4-Wiki #57: “Retry” and “Start over” options for SCREWS_TILT_CALCULATE

mrelwood, 2025-04-21:
The process starts with reverse homing to level the tilt of the bed. While this is an important step to do before tramming, it’s not precise enough to be done over and over again during subsequent retries. As a result, the adjustment values can vary as much as +/- 15 units even when only running the process again without making any adjustments.

The best solution I can think of are offering two options after having run the Calculate: “Retry” and “Start over”.

“Start over” would start the whole process from the beginning and include reverse homing, just like the current option does.

“Retry” would restart the process WITHOUT reverse homing.

I think it’s safe to assume that the bed frame tilt hasn’t changed enough during adjusting the screws to warrant new reverse homing that would just make the user run after a moving target when adjusting the screws.

Additionally I wish that the movement speeds would be more reasonable. The X and Y axis movements could be four times as fast as they currently are, and I think the Z axis could be doubled as well during reverse homing (for example fast movement to 260mm, then slow to the end.)

avdudchenko, 2025-04-22:
I genearly have found the screws tilt calculate to be a nightmare to use. It seems to be impossible to get the bed level with in 200 microns across four corners. If you do get it with 50 microns,, the next time you heat/cycle the bed it will be out of 200 microns again and the screw_tilt_calculate will complain. 

I think the reason we want to leave the homing operation is it might be possible to twist the bed during the adjustment process. You do need to apply a bit of force/torque to break loose the lock nuts and twist adjustment screwes etc.  

The fundamental issue to me seems the leveling blocks are not precise enough to retain level with in 10-50 microns) furthermore, expasnon of bed/screws/rubber supports varies enough to throw bed out of level by 100/200 microns.  

So in general their issue is the tilt_screw_calculate threshold for whats level is overly optimistic and will make you waste a ton of time...

The 100-200 microns is not even that bad considering the size of the bed...

avdudchenko, 2025-04-22:
A follow on question, If we are always fighting left/right leveling, why do we not have a macro to simply "level" left right balance useign the two z - motors. 

E.g. in theory we should only have to level bed front to back, and the left right should be correctable using the two z-motors. This would really simplify the whole process..... (I will need to look  into this).

TheRealBeef, 2025-05-22:
I think using the Z_tilt_adjust when doing the screws adjustment is a (potentially) bad idea, as this adjustment will happen on print_start and take out any slack

The idea of the screws adjustment is just to get the bed reasonably close so that bed meshing doesn't have to make overly large corrections

However, I can imagine a situation where a very crooked bed (such as the one that's out be >1mm I saw on the Discord yesterday) that will rack the bed sideways in order to be level and therefore won't show up in the screws tilt calculate macro.

The biggest sources of being off are
1. The bottom of the frame itself has 4 holes that the "legs" of the printer bed fit into, if there is junk in these it can push one of the legs upwards or downwards causing failed repeatability. This is why some recommend putting hardstops that raise this up slightly and give more repeatable results
2. There is some inherent error in homing against the bottom of the bed, as it may not be level enough even with such hardstops to be perfect. However, in my own experience I'm getting results that are generally not varying +/- 0.01mm with the occasional +/- 0.08mm run. One of the things that helps I think is tapping the bed after making the screw adjustments to try and eliminated any wiggle. Perhaps instead of adding the screws_tilt_calculated adding the vibrate macro that shakes the bed would do the same thing

avdudchenko, 2025-06-09:
Z_tilt_adjust  uses inductive probes to level left/right ,as such we do not need to worry about frame warp/or bottom stops etc. 

I found that it helps a lot in speeding up how bet leaving, as you only need to do front to back level. 

My issue still is that no matter how I tweak the induction probes I get poor reads. I need to make an issue on this as its unrelated.

closed 2025-11-08 by vega-d