qidi-community/Plus4-Wiki #60: Can beacon safely disconnected during print without causing mcu shutdown error?

rhiz0matic, 2025-04-30:
The beacon rev-h is rated for 115c. I've had a couple of instances where I'm printing very slowly on a print bed that is 120c and the beacon overheats and disconnects, causing the printer to fault and request to reset the firmware once the mcu-beacon is reconnected (I just wait a minute or two to cool down and can resume the print if I shut off the switch and back on, so the printer thinks it's a power failure) 

To avoid this. Is there a way to "disconnect" the beacon from the required mcu inputs during printing from the printer so that the printer will still run if it loses connection after the bed meshing is already complete and printing has begun?

stew675, 2025-05-01:
This would be a question better directed to the Beacon support staff.  As far as I am aware the Beacon unit is not able to be disconnected during a print, but they may have a solution.

The other suggestion would be to ensure that there is at least some small amount of air-flow over the beacon unit.  While the print bed may be 120C, the air in the chamber certainly is not, and it would not take very much airflow at all to ensure that the Beacon does not reach >115C

linouxis9, 2025-05-06:
@rhiz0matic Kalico (Danger Klipper) supports setting some MCU as optional (https://github.com/KalicoCrew/kalico/pull/339), with it, you could set the Beacon's MCU as optional / non-critical. If you know a bit of Python, you can try to backport this pull request to your Plus 4's klipper installation.

closed 2025-05-20 by rhiz0matic