qidi-community/Plus4-Wiki #61: Chamber Temperature Investigation and v1.6.0 Firmware

TheRealBeef, 2025-05-02:
https://github.com/qidi-community/Plus4-Wiki/blob/main/content/chamber-heater-investigation/README.md

Following the instructions here, I wanted to test the chamber heater to see if it functioned properly before doing a print, but seemingly no matter what I do the chamber heater refuses to turn on. I have the following in my printer.cfg - the v1.6.0 config is commented out in favor of the config presented. After this failed to work, I tried adding the following 3 lines from the original config, but this also didn't improve it.

`target_max_temp:65`
`heat_with_heater_bed:True`
`heat_with_heater_bed_tem_add:25`

Is there something that I'm missing?

```
# [heater_generic chamber]
# heater_pin:U_1:PC8
# max_power:0.4
# sensor_type:NTC 100K MGB18-104F39050L32
# sensor_pin:U_1:PA1

# control = pid
# pid_Kp=63.418 
# pid_Ki=1.342 
# pid_Kd=749.125

# min_temp:-100
# max_temp:70
# target_min_temp:45
# target_max_temp:65
# heat_with_heater_bed:True
# heat_with_heater_bed_tem_add:25

[temperature_sensor chamber_probe]
sensor_type:NTC 100K MGB18-104F39050L32
sensor_pin:U_1:PA1

[heater_generic chamber]
heater_pin:U_1:PC8
max_power:0.4
control = pid
pid_Kp=63.418
pid_Ki=1.342
pid_Kd=749.125
min_temp:-100
max_temp:80
target_min_temp:10
target_max_temp:65
sensor_type: temperature_combined
sensor_list: temperature_sensor GD32, temperature_sensor chamber_probe, temperature_sensor GD32, temperature_sensor GD32
combination_method: mean
maximum_deviation: 70
heat_with_heater_bed:True
heat_with_heater_bed_tem_add:25
```

closed 2025-05-21 by TheRealBeef

rhiz0matic, 2025-05-21:
just curious what your solution was?

TheRealBeef, 2025-05-22:
At the time (with new printer) I wasn't aware that Qidi disables the heater unless the bed it at its setpoint and >70c. 

So I just wasn't familiar with how the system was intended to function.